AronHetLam/ATEM_tally_light_with_ESP8266 #83: Tally Light Setup not making changes

WesJacksonDangIt, 2023-05-23:
I've set up several of these before but our network changed and now I'm running into an issue I've never had before. 

Everything seems to run smoothly loading the code and can access the Tally Light setup UI.
When I make the changes to Tally Light name, number, network name, et cet... and hit save, it doesn't save any of those changes. I can refresh the page and it goes back to the default settings, or if it's one that had previous settings they stay the same. 

The Connection Status at the top of the screen says "Changing State..." which I don't believe I've seen before. Any help would be great.

dmitrylive, 2023-06-01:
i have the same problem.
D1 mini pro 4mb. 
only wifi settings are saved.

```
In file included from /Users/grigoryandmitry/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.2/libraries/Ethernet/src/EthernetUdp.h:37,
                 from /Users/grigoryandmitry/Documents/Arduino/libraries/ATEMmin/ATEMmin.h:35,
                 from /Users/grigoryandmitry/Downloads/ATEM_tally_light_with_ESP8266-3.2.0/ATEM_tally_light/ATEM_tally_light.ino:35:
/Users/grigoryandmitry/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.2/libraries/Ethernet/src/Ethernet.h:39: warning: "MAX_SOCK_NUM" redefined
   39 | #define MAX_SOCK_NUM 8
      | 
In file included from /Users/grigoryandmitry/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.2/libraries/ESP8266WiFi/src/ESP8266WiFi.h:28,
                 from /Users/grigoryandmitry/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.2/libraries/ESP8266WebServer/src/ESP8266WebServer.h:30,
                 from /Users/grigoryandmitry/Downloads/ATEM_tally_light_with_ESP8266-3.2.0/ATEM_tally_light/ATEM_tally_light.ino:30:
/Users/grigoryandmitry/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.2/cores/esp8266/wl_definitions.h:42: note: this is the location of the previous definition
   42 | #define MAX_SOCK_NUM  4
      | 
In file included from /Users/grigoryandmitry/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.2/libraries/Ethernet/src/EthernetUdp.h:37,
                 from /Users/grigoryandmitry/Documents/Arduino/libraries/ATEMmin/ATEMmin.h:35,
                 from /Users/grigoryandmitry/Downloads/ATEM_tally_light_with_ESP8266-3.2.0/ATEM_tally_light/ATEM_tally_light.ino:35:
/Users/grigoryandmitry/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.2/libraries/Ethernet/src/Ethernet.h:150: warning: "UDP_TX_PACKET_MAX_SIZE" redefined
  150 | #define UDP_TX_PACKET_MAX_SIZE 24
      | 
In file included from /Users/grigoryandmitry/Documents/Arduino/libraries/ATEMbase/ATEMbase.h:36,
                 from /Users/grigoryandmitry/Documents/Arduino/libraries/ATEMmin/ATEMmin.h:34,
                 from /Users/grigoryandmitry/Downloads/ATEM_tally_light_with_ESP8266-3.2.0/ATEM_tally_light/ATEM_tally_light.ino:35:
/Users/grigoryandmitry/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.2/libraries/ESP8266WiFi/src/WifiUDP.h:28: note: this is the location of the previous definition
   28 | #define UDP_TX_PACKET_MAX_SIZE 8192
      | 
In file included from /Users/grigoryandmitry/Documents/Arduino/libraries/FastLED/src/FastLED.h:75,
                 from /Users/grigoryandmitry/Downloads/ATEM_tally_light_with_ESP8266-3.2.0/ATEM_tally_light/ATEM_tally_light.ino:37:
/Users/grigoryandmitry/Documents/Arduino/libraries/FastLED/src/fastspi.h:157:23: note: '#pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output'
  157 | #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
. Variables and constants in RAM (global, static), used 37684 / 80192 bytes (46%)
║   SEGMENT  BYTES    DESCRIPTION
╠══ DATA     1504     initialized variables
╠══ RODATA   9396     constants       
╚══ BSS      26784    zeroed variables
. Instruction RAM (IRAM_ATTR, ICACHE_RAM_ATTR), used 61303 / 65536 bytes (93%)
║   SEGMENT  BYTES    DESCRIPTION
╠══ ICACHE   32768    reserved space for flash instruction cache
╚══ IRAM     28535    code in IRAM    
. Code in flash (default, ICACHE_FLASH_ATTR), used 289040 / 1048576 bytes (27%)
║   SEGMENT  BYTES    DESCRIPTION
╚══ IROM     289040   code in flash   
esptool.py v3.0
Serial port /dev/cu.wchusbserial14230
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 34:94:54:8e:14:43
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0240
Compressed 332624 bytes to 237770...
Writing at 0x00000000... (6 %)
Writing at 0x00004000... (13 %)
Writing at 0x00008000... (20 %)
Writing at 0x0000c000... (26 %)
Writing at 0x00010000... (33 %)
Writing at 0x00014000... (40 %)
Writing at 0x00018000... (46 %)
Writing at 0x0001c000... (53 %)
Writing at 0x00020000... (60 %)
Writing at 0x00024000... (66 %)
Writing at 0x00028000... (73 %)
Writing at 0x0002c000... (80 %)
Writing at 0x00030000... (86 %)
Writing at 0x00034000... (93 %)
Writing at 0x00038000... (100 %)
Wrote 332624 bytes (237770 compressed) at 0x00000000 in 5.7 seconds (effective 466.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

AronHetLam, 2023-06-01:
Sorry for not getting back to you @WesJacksonDangIt, I've been busy lately.
However, I'm glad you found the issue you had @dmitrylive.
Does this solve it for you as well @WesJacksonDangIt?

AronHetLam, 2023-09-03:
I'll close this as i consider it stale by now 🙂

closed 2023-09-03 by AronHetLam