AronHetLam/ATEM_tally_light_with_ESP8266 #86: Problem with a design mod

Jhalfast, 2023-08-31:
I've never programmed an ESP chip before - and here's my use-case:

I'm designing a tally system that triggers some el wire instead of a LED, by sending the ESP8266 signal to the el wire's inverter - one each for R-G-B. Test on the el wire shows it works in this manner (applying direct power to it at least up to 5v), and by all appearances the esp works as expected (blue LED on the board is on, unless the tally is supposed to be red, when it turns off)... but I get no voltage readings at all from the board when I test it - with or without the el wire inverters.

If I'm not mistaken, the resistors indicated in this project are for the LED's sake, not the ESP's sake - so why won't a direct 3.3v signal normally thrown at the LED work as power to the inverters? Why can I not get any voltage readings from the ESP's when they are functioning and responding to the ATEM?

Jhalfast, 2023-08-31:
See picture. The el wire  inverters come from inside the 2 AA battery cases of cheap Amazon packages. Stats say these little boards pull about 10mA at max, the esp8266 stats say it tolerates up to 12mA ...so if anything is grossly out of tolerance, I can see that it could have fried the pins. I'll load up a brand new chip without the led or inverters, and measure - see if it reads the right voltage then. I followed the instructions as far as selecting the "LOLIN(WEMOS) D1 R2 & mini board", but all I could find out about my little board was that it was "100% D1 compatible"

This experiment is actually replacing a very unreliable Pi Zero system, working off MOSFETs to get the Pi signal to trigger a 12v LED system. I was hoping to avoid the added complexity by using this system, but I'm ready to go back there if I have to.

[image]

AronHetLam, 2023-09-01:
Looks nice. I hope you get it working!

It looks like a D1 mini, and for that you selected the correct board. I'll await your measurement on a new board.

What happens if you connect the EL wire to 3.3v on the D1 mini directly. The 3.3v pin gets power prom the boards power supply,  and not the ESP8266 chip itself, as it is what powers the ESP. It can supply more current.

Something else that could cause an issue is that the pins are outputting a PWM signal if the LED brightness setting isn't set to 255 (max).

AronHetLam, 2023-09-06:
You could probably just use the MOSFETs for the EL wire if they work with 3.3v

Jhalfast, 2023-09-06:
That’s true, and I tested the wire on 5v direct as well as the 3v AA’s, I could just run the power supply straight to them and skip the step-up (or down). It got hot (and brighter!) ...well, warm - not hot. With the nature of the system, they won’t be powered for very long anyway… If they last for a year, it still might be worth it.

Thanks for that - I just stopped thinking about it when I saw how dim the wire was.

His,

Jamey Halfast
Production Director
Glen Meadows Baptist Church
...making disciples who make disciples

AronHetLam, 2023-09-07:
Is it still dimmer if you use the 3.3v pin on the D1 mini than with 2 AA batteries? They'd probably last longer, but at least you know why the inverters might fail 😂

I'd love to see a picture once it's mounted, it would interesting to see it in action.

Funny, I'm currently a part of production in a baptist church as well.

Jhalfast, 2023-09-07:
I will certainly do this, when I get back to it. Church production is my full-time, and most of my projects stop at “it works” and I move on. Right now, I’ve simply taped up the ESP’s and about to do the same to the breadboards that have been permanent since I got the Pi version working! I even bought a 3d printer to make some cases, but I have never been able to get that far. It’ll look professional …one day.

His,

Jamey Halfast
Production Director
Glen Meadows Baptist Church
...making disciples who make disciples

AronHetLam, 2023-09-07:
We're not that big, so all volunteers 🙂
If you happen to have PTZ cameras that support Visca over IP, I might have something interesting.
I made [this](https://github.com/AronHetLam/PTZJoystickControl) application that lets you control PTZ cameras with any game controller that connects to a computer. Currently only the released Windows installer works, but the application compiles and runs on Mac, Windows and Linux (also a PI)

Jhalfast, 2023-09-07:
We will actually be adding a PTZ in short order, but what I am shopping for is a programmable one - so I can execute a slow pan-zoom across the stage, and it land beautifully and not just stop like a machine. That, and the ability to follow the subject. Those two things are most important for me right now, although your controller would definitely bring in new life from the traditional joystick!