AronHetLam/ATEM_tally_light_with_ESP8266 #87: So many d1 variants... (bunch of different questions)

SamVimes78, 2023-09-02:
Searching aliexpress for d1 minis right now. There are so many variants. I'm looking for a 5-pack with usb c. There seems to be a "4.0.0" variant with usb c that has a different naming scheme for the pins and also has an unpopulated back (no metal housing thingy).

There are also some that look just like the one in your picture, just with a usb c connector. But their description is unclear about flash memory (1mb or 4mb).

Afaik the aliexpress knock-offs work just like the wemos originals.

And then there are also the s2 esp32 boards...

How important is the flash memory size? Do you expect any problems adjusting the code for a different naming scheme? How complicated can it be to adjust the code to an s2 board?

I already built some tally lights with the m5stickc. They work just fine but everything is hardcoded. Want to give your variant a try because i'd like to help out a friend by building cheap tally lights for him that can be configured without having to reprogram them.

One thing i don't understand: Can i use just any 8266 board or does the d1 have a unique feature needed for this project? I also didn't know that you can configure settings with these things that are remembered even after a restart. Or do you have to configure them every time you want to use them???

AronHetLam, 2023-09-02:
Any ESP8266 should be fine, but the 1mb versions are most likely too small (I haven't tested though).
The only thing is that the pins would be different on other models. You'd need to modify the code accordingly, but that shouldn't be too hard. If you have at least 7 gpio pins you can just replace pin numbers/names in the top of the file. If you have less you'd need to remove some code for pins you don't need.

It can also run on an ESP32, but again the pins would most likely need to be modified. This also means the M5stickC should be able to run it, but you'd need to implement your own display code, and probably remove some LED code. You might be able to steal some display code from the project you're currently using.

Whichever module you choose, just make sure it's supported by an Arduino boards manager, and select the correct board when programming.

The settings in the configuration will be stored across reboots.

SamVimes78, 2023-09-02:
> The settings in the configuration will be stored across reboots.

Only across reboots or even when disconnected from a power sources? That's news for me!

AronHetLam, 2023-09-02:
Also when disconnected from power 😊

AronHetLam, 2023-09-03:
I'll close this as I believe your questions are answered, but feel free to write here again if anything comes up.

closed 2023-09-03 by AronHetLam