AronHetLam/ATEM_tally_light_with_ESP8266 #92: Feature Request: Pre-compiled binaries

BenJamesAndo, 2023-09-14:
Hello. I just discovered this project and I'm super happy with the way it implements tally features.

I wonder in order to make it a little easier for new people, would it be possible to supply pre-compiled .bin files for ESP8266 and maybe even ESP32 boards? That way for those following your instructions, instead of downloading Arduino IDE, adding in all the files, compiling etc. they could just download the .bin file and then flash it using [esphome-flasher GUI tool](https://github.com/esphome/esphome-flasher).

AronHetLam, 2023-09-15:
I'm glad you like it. :)
That would definitely make it easier for a lot of people. I see that [ESP web tools](https://github.com/esphome/esp-web-tools) is probably what we should go for, but a few things would need to be done before being able to get there.

If you're able to help that would probably make it go faster, as I'm limited on time at the moment.
However, I do have a concern regarding pinouts on different boards. Do you have experience with creating generalized binaries for the two platforms, and if there is anything else that needs to be considered?

This is initially what i see needs to be done:
- [ ] Generalize GPIO pins
- [ ] Build pipeline(s) to publish binaries
- [ ] Set up [ESP web tools](https://github.com/esphome/esp-web-tools) on [GitHub pages](https://aronhetlam.github.io/ATEM_tally_light_with_ESP8266/) page (Probably needs a separate DIY guide page.
- [ ] Update README, Wiki and DIY guide

BenJamesAndo, 2023-09-18:
Yes ESP web tools would definitely be the easiest. I've used WLED before and they use ESP web tools. It's super easy to install and then you simply plug it in again and visit the same page to update the firmware or change WiFi credentials.

I have 3 [ATOM Matrix ESP32](https://shop.m5stack.com/products/atom-lite-esp32-development-kit) boards and one ATOM Lite (I think it's a lite but not fully sure). I bought a fully assembled ESP32 tally light from https://www.etsy.com/au/listing/1501092899

So I'm not familiar with the ESP8266 boards.
I wonder if there was an option to allow changing the DATA_PIN via the HTML page then it'd put less need on getting the GPIO pins perfect, as there so many different versions of boards anyway?
For the MATRIX I put DATA_PIN 27 and the ATOM Lite as 25.
To get it to compile I had to change [line 476 ](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/blob/96938cc28073d9efbcf8f92c0b160d30bd881a2c/ATEM_tally_light/ATEM_tally_light.ino#L476) to `digitalWrite(pin, value);`
(I can open a separate issue about this).

I'm happy to help with testing but unfortunately I don't have any experience with building pipelines or setting up ESP web tools.
Perhaps a first step would be to compile some binaries, then update the guide to mention the ESP Home GUI flasher that you can download. Despite it being deprecated it still works really well for the time being.
Then once that's ready and binaries are working for people, we could explore setting up ESP web tools?

I can provide the MATRIX binary if you like.
I'd also be happy to sponsor this project to allow enough funds to buy a M5 Matrix if that's something you'd like.
There's a great [3D print file](https://github.com/josephdadams/TallyArbiter/tree/master/listener_clients/M5AtomMatrix-listener/m5_atom_mount) for it.

Additionally, some useful tools that could go in the Wiki or README that can help with this project is the [AtemProxy](https://github.com/LibAtem/AtemProxy) which is a command line program that can essentially act as an Atem server for the tally lights. The benefit of it is that you can have way more than 5 devices connected, and in my testing has no noticeable delay. Whereas running a tally light as a server using this firmware I noticed a slight delay.
The [pyAtemSim](https://github.com/jonknoll/pyAtemSim) was also useful for me to test the tally lights at home where I don't have an Atem mixer. Was easier than running ATEM_tally_test_server

AronHetLam, 2023-09-21:
I couldn't help myself, and have spent way too much time on this, but i migrated it over to use PlatformIO, and made a pipeline that builds the binaries and publishes them with ESP web tools [here](https://aronhetlam.github.io/ATEM_tally_light_with_ESP8266/install.html). It currently supports ESP8266 and ESP32 with all the pins still being hardcoded.
But i made it so that the remaining ESP32 variants should be easy to add.
It's currently all in __platformio__ branch.

Making the pins configurable is probably the way to go, instead of having a binary for each board. However, it could lead to strange behavior if the wrong pins are selected, or the same pins are assigned multiple times. We also need to make sure it's initially backwards compatible, so that people updating don't need to change the config or re-solder.
If pin config is something you want to take a crack at, it would be a great help.

> To get it to compile I had to change [line 476 ](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/blob/96938cc28073d9efbcf8f92c0b160d30bd881a2c/ATEM_tally_light/ATEM_tally_light.ino#L476) to `digitalWrite(pin, value);`

I don't have problems with this when compiling for my ESP32. What's your config? I use `ESP32 dev moule`.
I made a workaround for ESP32 not supporting `analogWrite()`, so that line should never be reached when compiling for ESP32.

> I'd also be happy to sponsor this project to allow enough funds to buy a M5 Matrix if that's something you'd like.

Funds are nice, and extra gadgets are always fun as well. Shipping to Denmark is probably more expensive than the M5 Matrix itself. It'll be around $40-50 including shipping to get it here. If you're still interested, the easiest is probably [PayPal](https://www.paypal.me/aronhetlam)

I will try to remember listing the tools once i get to updating the documentation.

BenJamesAndo, 2023-09-21:
Wow man, this is amazing! I successfully installed it on my M5 Atom Matrix via ESP web tools. Very cool indeed.
Obviously without being able to change the pins I can't use it just yet. But very handy for those with the two supported boards.

How many different boards/pin layouts do you think people would have? Now that you've made the pipelines perhaps adding in a few more ESP32 variants is the way to go for now. That way it's configured correctly as soon as it's installed if they're using a supported board.

Seems the most popular ESP32 boards that require no soldering and work well for tally's are the M5 Atom Matrix and the M5StickC. My understanding is the ESP web tools should be able to distinguish the difference between the two and then flash it with the appropriate binary.
Then if I figure out how to add the code to allow changing the pins via web ui then those using a different, or modified board can change it.

Currency conversion from my currency (AUD) to yours was super steep on PayPal, so I've just bought you some coffees instead.

AronHetLam, 2023-09-22:
The problem is that it's only possible to distinguish between chip models - not the boards they are on.

> Manifests describe the firmware that you want to offer the user to install. It allows specifying different builds for the different types of ESP devices. Current supported chip families are ESP8266, ESP32, ESP32-C3, ESP32-S2 and ESP32-S3

https://esphome.github.io/esp-web-tools/

What i meant was that i don't have binaries for the other ESP32 chip models, but that those would be simple to add with config.

If i have to publish a binary for each board model, there needs to be an install button for each of them or a drop-down to manually select it.

And thanks for the ☕!
I'll see if a can get an M5 Matrix my way 😊

AronHetLam, 2023-10-13:
I got the Matrix now, and have made a binary for it. It's on the web installer.

AronHetLam, 2023-10-13:
I Also Added [Improv](https://www.improv-wifi.com), which lets you set WiFi credentials from the web installer, which is pretty neat.

BenJamesAndo, 2023-10-16:
Very nice. I successfully flashed my Matrix device. Super easy!