AronHetLam/ATEM_tally_light_with_ESP8266 #96: Slow Response Time

JCrankWV, 2023-11-01:
Hello and thanks for the awesome code,

I have this setup using a 4mb DI Mini with 1 neopixel for the LED.

My problem is I'm seeing a 3-7 second delay in the LED changing when switching inputs.  My switcher is a ATEM 2 M/E Production Studio 4k.  Is this a normal delay?  Do you have any suggestions for making it quicker?

The wifi is Unifi.
The wifi signal on the ESP Tally shows -61 db

Thank you,

AronHetLam, 2023-11-02:
Hi, you're welcome.

I assume you're using the web-installer?

Your delay is high, the change should be instant (almost).
Do you know if you're on a busy network? If there is a lot of UDP broadcast data it might overwhelm the D1 mini.
There is some network discussion in #53 that might be helpful.

You could try to make a test with the tally test server program as fake swithcer, to see if it's still slow on a more isolated network. Once the test server code is installed you can select the `logs & console` option in the web installer, and press `enter` in the console to switch between tally states.

Alternatively you could try the more powerful ESP32. If you don't want any soldering there is the [Atom matrix](https://shop.m5stack.com/products/atom-matrix-esp32-development-kit)

JCrankWV, 2023-12-21:
Thank you for your response,  Sorry it took so long for me to respond.  Been a bit busy.

Yes I used the web installer.  Is the arduino editor a better route?
The network should not be that busy, but I don't know how much broadcast traffic there is.  We are running a Unify system and my production equipment is on its own VLAN.

I'm going to give an esp32 a try.  Will the web installer work for the ESP32?

AronHetLam, 2023-12-25:
The web installer is fine. For supported boards It shouldn't make a difference.
ESP32 is supported, but depending on which module you get the pin layout might not be ideal. To customize it you'd need the Arduino IDE (simpler) or PlatformIO (more advanced)