AronHetLam/ATEM_tally_light_with_ESP8266 #97: improv.setDeviceInfo(CHIP_FAMILY, DISPLAY_NAME, VERSION, "Tally Light", "CAM1");

szdn1, 2023-11-04:
Hi
error problem

In file included from c:\Users\ASUS\Documents\Arduino\libraries\FastLED\src/FastLED.h:75,
                 from C:\Users\ASUS\Documents\Arduino\libraries\ATEM_tally_light\ATEM_tally_light.cpp:49:
c:\Users\ASUS\Documents\Arduino\libraries\FastLED\src/fastspi.h:157:23: note: '#pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output'
  157 | #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\ASUS\Documents\Arduino\libraries\ATEM_tally_light\ATEM_tally_light.cpp: In function 'void setup()':
C:\Users\ASUS\Documents\Arduino\libraries\ATEM_tally_light\ATEM_tally_light.cpp:294:26: error: 'CHIP_FAMILY' was not declared in this scope
  294 |     improv.setDeviceInfo(CHIP_FAMILY, DISPLAY_NAME, VERSION, "Tally Light", "CAM1");
      |                          ^~~~~~~~~~~

exit status 1

Compilation error: 'CHIP_FAMILY' was not declared in this scope

Please Help me

szdn1, 2023-11-04:
use D1 mini pro

AronHetLam, 2023-11-04:
I just pushed a fix for that a couple of hours ago. It's easiest if you use the web installer, but the latest code on the main branch should work in the Arduino IDE. Make sure to copy the libraries over again.

AronHetLam, 2023-11-04:
Just created Release `v3.3.1`. That should do the trick, and make the download links point to the fixed version. 🙂

closed 2023-11-04 by AronHetLam

szdn1, 2023-11-04:
Thank you very much, it was fixed
you are the best