AronHetLam/ATEM_tally_light_with_ESP8266 #98: ISO Rec mode (akin to On Air mode)

Gracham, 2023-11-21:
Hi there
For starters I would like to say that I find this project very impressive and I would like to thank you for your work Aron!
I'be been using it for some time as a primary tally light system for my live streams.

Recently I have found myself in need of indicating ISO recording status by a light placed in front of the doors to the studio.
I'm not familiar with ATEMLib or creating new library entries but I haven't found anything that might be able to get the recording status in current library included in the repo.

I would love to see an ISO Rec mode akin to the On Air mode in which tally light lights up red when Atem is ISO recording.
If not in the official project, I would be greatful for tips in terms of modification of existing code to add that functionality!

Thank you in advance

AronHetLam, 2023-11-21:
Hi, I'm glad you're liking it. 🙂
But be warned - you're going straight for the deep end!

It should be possible to add, but there's no official library. It's all based on reverse engineering by Skaaroj, and he sadly stopped publishing updates years ago.

I added On Air myself in my copy by coaming through UDP packets with wireshark. Essentially you'd need to do the same.
Maybe the Companion module for ATEM switchers has it implemented. If so, it's probably way simpler to look at their code for the right command, and add it here. 

Sadly I'm limited on time at the moment, but if you find it I might be able to implement it, or show you where to look.

Gracham, 2023-11-21:
Thank you for your swift response!
I completely understand!
I've found some actions related to Recording in [companion-module-bmd-atem/src/actions.ts](https://github.com/bitfocus/companion-module-bmd-atem/blob/master/src/actions.ts)
Please correct me if i'm looking at the wrong thing

[image]

[image]

AronHetLam, 2023-11-21:
You're on the right path I think, but you have to get more low level.

Here you see what I added to parse raw streaming status data. https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/blob/master/libraries/ATEMmin/ATEMmin.cpp#L344-L361
You need to find a four letter string corresponding to ISO recording, and what the following data means.

Now that I'm thinking about it, maybe some recording commands can be found in Skaarhojs other ATEM libraries. He had different sizes with support different commands, so that you could run a smaller one on smaller hardware. Not sure if ISO recording is there.
My copy is the smallest one.

Gracham, 2023-11-21:
Thanks for the tips!
I can't find anything deeper related to Recording in bdm-atem repo and -base repo. I'm probably looking in wrong places as im not familiar with that codebase. I'll try looking further but i doubt my code skills.

As a workaround I could try replacing streaming state with one of the streaming error states you've implemented. That way by not filling out stream key I could use on air button and mode to turn on the red light while not putting load on the switcher needlessly

AronHetLam, 2023-11-21:
Maybe that could work.

Else this might. https://github.com/nrkno/sofie-atem-connection/tree/master/src%2Fcommands%2FRecording

Gracham, 2023-11-24:
Hi there, I've snooped around the code examples and other Atem libraries but im not code savvy enough to get useful serial data out of it.
I've managed to find packet that I think corresponds to starting a recording via Wireshark

0000   18 7e b9 07 f6 93 7c 2e 0d 16 3e 9c 08 00 45 00
0010   01 30 bf 1f 00 00 ff 11 79 1b c0 a8 00 90 c0 a8
0020   00 a1 26 b6 e3 74 01 1c e0 f0 89 14 80 82 6d 8f
0030   00 00 00 00 0f 4d 00 10 00 00 54 69 6d 65 0a 36
0040   2b 00 00 00 00 54 00 10 4f 00 52 54 4d 53 00 02
0050   35 31 00 00 16 79 00 54 53 54 52 54 4d 44 00 00
0060   00 00 00 00 16 7a 00 04 41 2d 49 53 4f 2d 44 79
0070   73 6b 31 00 53 44 53 52 53 44 00 00 00 00 00 00
0080   00 00 00 10 00 07 53 52 53 54 00 00 00 00 00 cc
0090   00 0b 00 10 17 70 53 52 53 44 00 00 00 00 00 00
00a0   00 00 00 02 f9 00 00 00 16 7b 00 10 1f 42 52 54
00b0   4d 53 00 02 00 23 00 00 16 79 00 10 41 2d 52 54
00c0   4d 53 00 02 00 00 00 00 16 7a 00 54 ff ff 52 54
00d0   4d 44 00 00 00 00 00 00 16 7a 00 08 41 2d 49 53
00e0   4f 2d 44 79 73 6b 31 00 16 79 00 04 41 2d 49 53
00f0   4f 2d 44 79 73 6b 31 00 ff ff ff ff 01 00 00 05
0100   00 ff ff ff ff ff ff 01 00 00 06 00 ff ff ff ff
0110   ff ff 01 00 00 07 00 ff ff ff ff ff ff 01 00 10
0120   08 00 52 54 4d 53 00 03 01 00 00 00 16 7a 00 10
0130   ff ff 52 54 4d 53 00 03 00 00 00 00 16 7a

Gracham, 2023-12-02:
I tried to replace "StRS" value in your streaming implementation with various 4 letter combinations from wireshark packets with no suceess. I figure it's not as straight forward as I suspected

Gracham, 2023-12-02:
Alright, found a solution! Code "TcLk" corresponds to both starting and stopping of a recording. 
When I eddited On Air option with TcLk value LED responded with ifstatement when command REC STOP was issued.
For now I've just swapped the LED colors to get behaviour close to what I want.
The only downside is that now the default state of LED is red, not green. Only when the recording starts and then stops it changes to green, but that can be easily fixed with some code.

[image]

Gracham, 2023-12-02:
Swapping Streaming Flag values resulted in LED reaction to REC START!
[image]

AronHetLam, 2023-12-02:
Sorry for the late reply, but i took a quick look at it.
Thanks for your captures. :)

I added the `RTMS` command, as that seems to be what the companion module uses, and also mainly looked at their definition of the status flags.
As each command has different flags just replacing the command as you tried to would lead to strange behavior, as the commands doesn't map one to one.

In [this](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/tree/feature/recording-status-command) banch I have added these functions to the `ATEMmin` library. 

```
uint16_t getRecordingStatusFlags();
bool getRecordingIdle();
bool getRecordingRecording();
bool getRecordingStopping();
bool getRecordingErrorNone();
bool getRecordingNoMedial();
bool getRecordingMediaFull();
bool getRecordingMediaError();
bool getRecordingMediaUnformatted();
bool getRecordingDroppingFrames();
bool getRecordingUnknownError();
```

It compiles, but I haven't tested it. Let me know if it works, and I'll merge it.

AronHetLam, 2023-12-03:
I found a bug.
Copy pasting is dangerous. I forgot to rename where I put the recording flag data, so it saved it to the `streamingStatusFlags` instead of `recordingStatusFlags`

AronHetLam, 2023-12-03:
Awesome 🙂

However, it'll probably be at least a few weeks before I get to adding it as an option.

Gracham, 2023-12-03:
My mistake, my account hit unlisted pastes limit and it went private automatically. It should be available now

AronHetLam, 2023-12-05:
`ATEM_min` uses `ATEM_base`, so that's not the issue.

Found that the library checks needs the passed value to be `0x80` instead of `0xff`, so
`atemSwitcher.serialOutput(0x80); //Makes Atem library print debug info`