AronHetLam/ATEM_tally_light_with_ESP8266 #103: Web UI settings are not saving

Jude741, 2023-12-11:
I'm using a D1 mini pro and i followed the instructions properly and i have reached til web interface setup. It connects to wifi but apart from ssid and password no other settings are saved. It says “Settings saved successfully” but the settings go back to the default values. Ex: Tally Light name ESP-93A6A8, static IP false and ATEM IP 0.0.0.0 etc.

Jude741, 2023-12-11:
I chose "LOLIN(WEMOS) D1 R2 & mini" in the boards manager instead of "D1 mini pro" and it worked. Thank you for your support.

AronHetLam, 2023-12-11:
Glad you got it working. 🙂

Did you install it from the web installer before?
If so, I need to figure out a fix for that.

AronHetLam, 2023-12-11:
If you didn't use the web installer, maybe I could have you test it as I don't have a D1 mini pro?

Jude741, 2023-12-17:
Sorry for my late response. I used the aduvino IDE to upload the program because i wanted to slightly modify. I haven't tried the web installer.

closed 2023-12-18 by AronHetLam