AronHetLam/ATEM_tally_light_with_ESP8266 #109: Not an issue

jmvb-cl, 2023-12-29:
Hello Aaron, hope you'r good

I updated my tallys with the new web installer (nice feature BTW).
Now, with the update done, when the tallys boot up, the red light turn on briefly, then no lights at all, until connect with the modem (the red light turn on briefly again) then off, or red if it's in program, or green if it's in preview, or purple if it's no connection.

Is this an expected behavior?

AronHetLam, 2023-12-31:
The colours shouldn't be different during the startup sequence, so it could sound like I need to look into why it's different.

jmvb-cl, 2023-12-31:
I'll make you a video

AronHetLam, 2024-01-21:
It shouldn't have changed, and i just fixed a small issue related to this. If you are able to verify if there are still differences to how it used to work, please let me know :)

jmvb-cl, 2024-03-13:
V3.3.2 still having no "BLUE" while connecting to WiFi

https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/assets/77258813/2630c6ce-0399-4980-9e2d-5e6b30c86ab1