AronHetLam/ATEM_tally_light_with_ESP8266 #110: Feature Request: WiFi Reset via a button (so we're back on the WIFI setup network)

MaierMatthias, 2024-01-10:
Hi there,

I really love the implementation and we're going to use it in a first production on the upcoming weekend. 

One thing which crossed our minds was if there's a button to completely reset the wifi settings and go back to providing the "Tally setup" WIFI network. I understand that in studio situations there's a fixed IP a convenient solution but sometimes networks on jobs change so we'd love to have the solution a bit more flexible.

I screened the code and one option might be setting `bool firstRun;` back to `true` but I wanted to ask first if there's a more elegant solution. I'm thinking of a logic like: "**when button X is pressed for >5 seconds, then reset the board and create the setup wifi network again**".

Thanks in advance! 

Best from Germany,
Matt

AronHetLam, 2024-01-13:
Hi, that's certainly possible to do. Making it just reset the WiFi credential using [this code](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/blob/master/ATEM_tally_light%2FATEM_tally_light.cpp#L955-L964) is probably the best way to do so.

However, if the tally lights can't connect to the specified network, it will automatically provide the "Tally Setup" WiFi again, so I'm not sure i understand why the button is needed?

If it's because you need to change settings even though the WiFi network is the same I would strongly recommend using static IP-Adresses in your setup.

chiflyer, 2024-02-17:
I have added an easy button function.  Feel free to add this to your wonderful and awesome script.  

/*
    Copyright (C) 2023 Aron N. Het Lam, aronhetlam@gmail.com

    This program makes an ESP8266 into a wireless tally light system for ATEM switchers,
    by using Kasper Skårhøj's (<https://skaarhoj.com>) ATEM client libraries for Arduino.

    This program is free software: you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation, either version 3 of the License, or
    (at your option) any later version.

    This program is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
    GNU General Public License for more details.

    You should have received a copy of the GNU General Public License
    along with this program.  If not, see <https://www.gnu.org/licenses/>.
*/

#include "ATEM_tally_light.hpp"

#ifndef VERSION
#define VERSION "dev"
#endif

// #define DEBUG_LED_STRIP
#define FASTLED_ALLOW_INTERRUPTS 0

#ifndef CHIP_FAMILY
#define CHIP_FAMILY "Unknown"
#endif

#ifndef VERSION
#define VERSION "Unknown"
#endif

#ifdef TALLY_TEST_SERVER
#define DISPLAY_NAME "Tally Test server"
#else
#define DISPLAY_NAME "Tally Light"
#endif

//Include libraries:
#if ESP32
#include <esp_wifi.h>
#include <WebServer.h>
#include <WiFi.h>
#else
#include <ESP8266WebServer.h>
#include <ESP8266WiFi.h>
#endif

#include <EEPROM.h>
#include <ATEMmin.h>
#include <TallyServer.h>
#include <FastLED.h>

#if ESP32
//Define LED1 color pins
#ifndef PIN_RED1
#define PIN_RED1   32
#endif
#ifndef PIN_GREEN1
#define PIN_GREEN1 33
#endif
#ifndef PIN_BLUE1
#define PIN_BLUE1  25
[... 977 lines cut ...]
//         //Set back status LED after one second to working LED_BLUE if it was changed by anything
//         if (lowLedOn) {
//             setStatusLED(LED_ORANGE);
//             lowLedOn = false;
//         }

//         //Blink every 5 seconds for one second if battery current is under 3.6V
//         if (lowLedCount >= 5 && uBatt <= 3.600) {
//             setStatusLED(LED_YELLOW);
//             lowLedOn = true;
//             lowLedCount = 0;
//         }
//         lowLedCount++;

//        //Turn stripes of and put ESP to deepsleep if battery is too low
//        if(uBatt <= 3.499) {
//            setSTRIP(LED_OFF);
//            setStatusLED(LED_OFF);
//            ESP.deepSleep(0, WAKE_NO_RFCAL);
//        }

//         secLoop = 0;
//     }
//     secLoop++;
// }