AronHetLam/ATEM_tally_light_with_ESP8266 #112: If the connection is lost, it does not clear the status of the LEDs.

szkrstms, 2024-01-20:
Hi,

I created a test server and a client. The client, on first startup, gives a nice indication that there is no connection(D1,blue).
But!
When I switch off the test server, only the terminal shows that the connection to the mixer is lost. Why does the D1(blue) LED not go off?
Here is a video of it: https://youtu.be/ZmzVYheIxJc
LED1 - D0(red)
LED2 - D1(blue)
LED3 - D2(green)
It's a bug, or I do something wrong?

Thank you

AronHetLam, 2024-01-20:
I don't think it's the code, but as I can't see the chip and wiring I can't confirm if you connected it correctly. However it looks like it could be wrong.
There are multiple states of connecting

Blue - connecting to WiFi
Pink (Both Blue and Red) - connecting to switcher

However, I only see either all LEDs being on, or the third only, in your video. 

Also the Blue led should not be able to turn on while connected, as it would only switch between red and green.

The board you have says vcc, indicating that's connected to the cathodes (+). The program is intended to have the pins connect to the cathodes (+), and the anodes (-) to ground. I think that might be what's causing your issue.

szkrstms, 2024-01-21:
Thank you for your reply, but my problem is not with the wiring of the LEDs.
Unfortunately, my English is very bad, I use a machine translator and I may have formulated my problem wrongly.
I made another video showing the spikes and wiring up close. https://youtu.be/zeMP9gWY4Mg
D0 pin is brown, D1 pin is black and D2 pin is orange. On the LED panel, LED1 is brown, LED2 is black and LED3 is orange.
That is, the LEDs indicate the status of pins D0, D1 and D2.

On power up, if there is a Wifi connection but it cannot connect to the mixer, the D1 pin indicates that there is a problem with the connection. It also displays the following 
"Connecting to the switch ...
Switch IP: 1912.168.0.208"
And here it waits until it can't connect.

After plugging in another device running the test server, it seems to reconnect. It says "Connected to switcher" and shows on the D1 output that the connection is OK.
As you see the test server control the D0 and D2 LEDs.

When I turn off the test server, it detects this and prints "Connection to switcher lost ...", but does not indicate with the LED on pin D1, and the status of legs D0, D2 is "stuck".
That is, it does not visually indicate that there is a problem with the connection.
Or is this just because I didn't connect 1 RGB LED to it, but I'm monitoring the status of the legs one by one?

AronHetLam, 2024-01-21:
You can use an individual LED for each pin, but the LED board you have cannot be used, without modifying the code.

The pins (D0, D1, D2) need to be connected to the cathodes/positive sides of the LEDs. On your board they're connected to the anodes/negative sides. This is most likely the cause of your issue. The Anodes/negative sides should be connected to ground.
 
As your board connects all the cathodes into one vcc pin, you won't be able to turn on individual LEDs by turning on the cathodes. With your board you need to do the opposite and turn off the pins you want an LED to turn on for.

AronHetLam, 2024-01-21:
Sorry, I swapped anodes and cathodes around. It should be correct now.
Anode is the positive side, so vcc.
Cathodes is the negative side, usually connected to ground.

AronHetLam, 2024-01-21:
.

szkrstms, 2024-01-21:
My problem is not with "active low" logic. I understand why the LED turns off when the program is on. This is how the test LED board is wired.
My problem is the D1 pin. When the connection is broken, it does nothing. It does not change the state to "alarm". It does not indicate the connection to the mixer is broken.
Watch the video again and look at the terminal messages and the D1 LED.

AronHetLam, 2024-01-21:
I must have missed that due to the strange LED behavior.
I found the issue and pushed a fix. The [web installer](https://aronhetlam.github.io/ATEM_tally_light_with_ESP8266/) should be updated as well.
I hope that solved it for you :)

closed 2024-01-21 by AronHetLam