AronHetLam/ATEM_tally_light_with_ESP8266 #115: Dry Contact Support

zdog16, 2024-02-16:
Could we add the ability to change the Output from an RGB LED to 3 dry Contacts?
I use this project on a Steadicam Rig and my [External Monitor](https://www.amazon.com/VILTROX-DC-70EX-1024x600-Monitor-Display/dp/B01LW59L7G/ref=sr_1_9?crid=3KPKKK08IGO1T&keywords=viltrox+monitor&qid=1708113990&sprefix=viltrox+moitor%2Caps%2C101&sr=8-9) has inputs for a built-in Talley Indicator. 
I'm assuming they're Dry Contacts, I haven't actually tested them yet, but Dry contacts should allow you to interface directly with the Talley indicators on a lot of cameras.

AronHetLam, 2024-02-16:
I'm not exactly sure what you mean, but yes you should be able to control your monitor's lights. You need to know what voltage they operate at, and get the right electronics, but essentially the RGB LED is just controlled by 3 individual pins being either on or off. The esp operates at 3.3v, so if you need more you need to get transistors or relays accordingly. https://a.co/d/4kM3Z21

zdog16, 2024-02-17:
A dry contact is just a connection where no voltage needs to be applied you only need to complete the circuit to activate the device at the other end. 

I did think about using relays like the ones you linked as the output side of a relay acts as a dry contact when no voltage is introduced. The issue I came up with is that since RGB LED uses a combination of signals to produce the color required. For Red and Green that's fine since those are 2 of the 3 colors in RGB but white would activate multiple pins (relays) and I'd probably have to do some fancy wiring to to close only a single relay, or re write the outputs portion of your code. 

I have just enough experience with coding in Arduino to figure it out eventually and make my own fork. However I figured this might be a feature others would find useful if they're in a similar situation using cameras that have Talley capabilities. Even if the board is not capable of outputting a "dry contact" an option in the web UI to change the output behavior to would be helpful.

AronHetLam, 2024-02-17:
I don't see why it would be a problem to turn on both the red and green lights on your monitor at the same time. You can just leave the blue pin disconnected. As soon as the esp is connected to the switcher only one will be on at a time unless it loses connection again.

Or am I missing something regarding your monitor? As I understand the plug for your monitors tally light it has 3 colours - yellow, red and green.

zdog16, 2024-02-17:
I haven't had a chance to power on the monitor yet, and the manual doesn't have any details about the Tally input. I would want to use the yellow light on the monitor to mimic the white status on the LED since the D1 Mini will be hidden away, and I'll want to know when the wifi is disconnected. While having multiple contacts closed on the input shouldn't hurt anything I suspect it will default to the red light when the D1 Mini closes all three relays.
I'm also not sure yet if the monitor has dedicated lights for the tally indicator or if it puts a colored border on the image. I know the Panasonic P2HD that I'm using puts a red border on the image in the viewfinder when tally is activated. That'll be my next project to hook one of the receivers into the camera directly so some of our other camera ops can also have tally, but the relays should work nicely for that.

I submitted the issue mainly for if others would find the feature useful, but if it doesn't align with the project goals feel free to close it. I love the project and I appreciate all the work you've done.

AronHetLam, 2024-02-17:
I see. The dry contact will require additional hardware no matter what, but a simple transistor is probably enough, assuming it only has to switch a low voltage signal.
When you get to test how the monitor actually handles the signalling, let me know. Maybe it makes sense to add a `Tally Only` and/or `Raw Tally`mode that will let the pins signal the raw tally state, and nothing else.

zdog16, 2024-02-19:
I finally got a chance to test the monitor today and I was correct, shorting the wires of a cut up Cat5 cable was all it took to turn on the Tally Indicators on the front face of the monitor above the screen. Interestingly there seem to be 3 discrete LEDs inside the each defuser as when I shorted all 3 pins the indicators lit up with 3 different colors. 
So I should be able to use the Tally as is. When it goes to "white" it'll light up all three LEDs which will tell me there's an issue. I Ordered some transistors and a protoboard so I can test it further.

jmvb-cl, 2024-03-12:
I use the Aron tally lights to turn on on-board Viltrox monitor tally light
I use the system to trigger an Arduino relays