AronHetLam/ATEM_tally_light_with_ESP8266 #122: Wt32-eth01 Support

jmvb-cl, 2024-03-25:
I'm looking add to network one wired tally unit which act like a server, in order to avoid Atem freeze

I found Wt32-eth01, which lack USB port, but I found a USB to UART TTL board to add, so, problem solved I guess

Did anyone use this board?

AronHetLam, 2024-03-26:
I haven't, but I'm interested to hear about the results if you try it out.

jmvb-cl, 2024-03-26:
What if instead buy you a coffee, buy you one of those? 
villeg.jm at gmail.com if you are interested..

jmvb-cl, 2024-05-23:
@AronHetLam you got it?

jmvb-cl, 2024-08-20:
I want to highlight the incredible, dedicated and selfless work of Aron
He has had the volition and patience to modify the code to make this device possible to work.
This, without counting the good disposition he had with me (and my poor computer skills) to ensure that I could program and operate the WT32
Problem solved!!

closed 2024-08-20 by jmvb-cl

jmvb-cl, 2026-08-20:
@AronHetLam ?