AronHetLam/ATEM_tally_light_with_ESP8266 #123: Suggestion for a Relay Shield Variant

W9TIM, 2024-03-26:
I have 5 of the LED units working together and using your guide and code it was very simple to implement. The addition of the 'On Air' option was a great improvement.

I would like to suggest a variant that doesn't use LED's but rather activates the Wemos Relay Shield so that a full size "On Air" light could be switched on and off following the state of the ATEM Tally or Streaming status. This would allow us to conserve the 4 Wifi ATEM connections while providing for control of other devices (lights) 

I have constructed one using the M5StickC-Plus and the M5Stick Relay module. But I think the Wemos D1 with Relay Shield could implemented for a much lower cost (<$10) and could be incorporated inside many available low voltage "On Air" signs available at a low cost online. 

I have the M5StickC running so that it activates the "On Air" sign when a live source & streaming are both active. This allows us to begin our streaming with a countdown screen or bumper clip with the light off and it only comes on when one of our 'live' sources is active in program mode. We use it to alert those conducting our services when they are "live on air". This works very effectively.

AronHetLam, 2024-03-26:
Did you Notice the server functionality? You can have one unit act as a server that the other units can connect to, in order to free up connections on the ATEM. That way you can still use ATEM Software control.

The relay shields should be fine to use. I can see it connects to D1 if you use it as a shield, which connects to a blue pin. You can either just have the relay connected to a red pin with a wire, og change the pin numbers in this part of the code https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/blob/master/ATEM_tally_light%2FATEM_tally_light.cpp#L82-L104
Just make sure not to use the same pin number twice.

I can't really change the pin numbers in a new release, as it would break functionality for others. I hope that makes sense 🙂

AronHetLam, 2024-03-26:
I see where you're going with this.
I'd much rather not add any more #ifdef and #ifndef precompiler statements, as it decreases readability by a lot, and already introduced bugs in earlier releases.

As the relay shield uses D1 you should be able to just make a new .ino file where you put the following statements:

```
#define PIN_RED1 D1
#define PIN_BLUE1 D0
```

That way the red pin will trigger the relay, and the rest of the pins are not connected, except for ground of cause.

W9TIM, 2024-03-26:
Yes, I think we are on the same page now.

There’s even a convenient enclosure .stl that includes the relay shield on Pintables. I appreciate your fine work on this project.

https://www.printables.com/model/41507-enclosure-for-wemos-d1-mini-with-the-relay-shield

W9TIM, 2024-04-01:
I worked out a way to test for more complex conditions that make the On Air Light relay option even more flexible. I've attached my .cpp file that contains the modification notes at the top. For my original goal using the 'Relay' version as the primary connection to the switcher and connecting the traditional tally lights to it, I both conserve connections and also provide the 'On Air Sign' functionality I was looking for.  - Your comments or further suggestions are welcome and appreciated. 

[ATEM_tally_light.cpp.zip](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/files/14818253/ATEM_tally_light.cpp.zip)

W9TIM, 2024-04-10:
I found a false on condition possible when booting up and ATEM is not active. That has been addressed in the file attached. 

Please note this modified version was constructed to support a RelayHat by re-configuring LED1, however, it can work seamlessly as a source for the basic versions as documented in the DIY guide. To add a relay-controlled light to their existing project or set of tally light modules. This code is only needed for the relay shield module. 

[ATEM_tally_light_Modified.zip](https://github.com/AronHetLam/ATEM_tally_light_with_ESP8266/files/14933474/ATEM_tally_light_Modified.zip)