AronHetLam/ATEM_tally_light_with_ESP8266 #128: Compilation error: FastLED.h: No such file or directory

Ibrohimofficial, 2024-04-27:
C:\Users\ROG Stix\Documents\Arduino\ATEM_tally_light\ATEM_tally_light.ino:37:10: fatal error: FastLED.h: No such file or directory
   37 | #include <FastLED.h>
      |          ^~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: FastLED.h: No such file or directory

I can't find three in one led than i use three red, green and blue leds maybe this error for it?

[image]

AronHetLam, 2024-04-27:
No, you've missed a step in the description for installing the fastLED library.
Your LEDs should be fine.

If you use the web installer you can program the chip from Google Chrome. You don't need the Arduino IDE at all, or to do any installation on your computer.

Ibrohimofficial, 2024-04-27:
Thank you very much, I forgot to install the library.

closed 2024-04-27 by AronHetLam