AronHetLam/ATEM_tally_light_with_ESP8266 #131: Could Not work with D1 Mini Pro

Budilla2, 2024-08-30:
HI All, i tried to use the tally with the d1 mini pro but it cannot be programmed, several chips same problem.

Is there a way to work with D1 mini pro? Has anyone tried and had success?

Thanks for your help
Regards

Marcelo
Brazil

jmvb-cl, 2024-08-30:
yes, I did success... how are you uploading the code?

Budilla2, 2024-08-30:
> yes, I did success... how are you uploading the code?

I tried to upload de code using the web installer exactly as i did with the D1 mini, but gives me a message as attached (failed to initialize). I have tried other 2 D1 mini pro and same problem. Maybe i'm doing something wrong.
[image]

jmvb-cl, 2024-08-31:
Seems like driver issue.. do you have parallels in your mac?

Budilla2, 2024-08-31:
Great! I tried on my son computer with Windows, i have instales the drivers
and boom! It worked!

Thanks for your support!

closed 2024-09-12 by AronHetLam