AronHetLam/ATEM_tally_light_with_ESP8266 #135: No COM option under Port

0h-Dang, 2024-10-11:
I hate to be a bother being a novice following a YouTube video, but in the portion of "Programming the D1 mini" under 'Tools -> port -> list of COM'

I don't see ANY COMs before or even after connecting the board to the computer. The options I see is (under Port) 
/dev/cu.BLTH
/dev/cu.Bluetooth-Incoming-Port
/dev/cu.usbserial-1440

maybe what I could've done wrong was on my first (1 of 3) boards I went through the "Flashing the firmware" via Web Installer https://aronhetlam.github.io/ATEM_tally_light_with_ESP8266/install and the light on the board went from solid red to a solid turquoise / teal blue. And after that I installed Arduino zip for the other two boards. Not sure how to continue from here.

nowhere in the walkthrough does it explain if we don't see COM under Ports.
[image]
[image]
[image]
[image]

AronHetLam, 2024-10-11:
No worreis
Are you on Mac? Then those three options you have are COM ports. I believe the last one is the ESP board, but to verify you can check the list before and after connecting the connecting the ESP. The one that only appears after would be the ESP.

If none of those are the board you might not have installed the driver linket in the guide.

AronHetLam, 2024-10-11:
By the way. Did the web installer not work for you?

0h-Dang, 2024-10-11:
Thanks for such a quick response. ill double check by connecting a board I haven't used yet.
I am using Mac, under driver initially didn't install it as it had a note that newer than Mojave didn't need to be installed (running macOS 14

after the installer I wasn't sure if I had to go straight to the networking stuff, or what color to expect. first time doing something like this.

AronHetLam, 2024-10-12:
It sounds like you did it correctly with the web installer. The turquoise you describe is probably "white", meaning it's unable to connect to WiFi. This would make sense while you haven't configured it. You can do so through the web installer by connecting the board to the installer again, or by following the next step in the guide.

0h-Dang, 2024-10-13:
when said white I expected bright white, threw me off. guess the web installer DID work! I just can't access the setup. connecting to "Tally Light Setup" but when I input the "192.168.4.1" but my browser can't reach it. either says "site can't be reached" or "Cannot connect to Server" But other than that your install process is soooo easy. thank you for that change!

0h-Dang, 2024-10-13:
maybe Macs see the unprotected network as an issue, might have to get a hold of a windows laptop?

AronHetLam, 2024-10-13:
Are you connected to another network as well? That could could make your Mac use that instead. Else try from a smartphone.
You should also be able to set WiFi credentials through the web installer by opening the installer again after you programmed the board. If that works, you can use the log through the web installer to see what it's ip address is.

0h-Dang, 2024-10-14:
One last thing, So I am trying to daisy chain 2 more boards to connect to my master light which is Cam 2 on my switcher. and I added the reserved IP of the master light as the ATEM IP on the other 2 boards. made the IP/s static but left the tally IP for those specific boards 255. all around. In your diagram it shows the (top) IP as grey so does that mean it shouldn't matter what it is? do I still need to reserve an IP for them?

connected all three to power and (main board) responds to preview / Live while the others stay Pink. looking at a table you have provided that means its trying to connect to the ATEM (in this case, Tally_Light 2" (Master)

any clarification for those IP on the other IP's. like if it matters what they are?

jmvb-cl, 2024-11-12:
In mac os, install the drivers and try Google Chrome.