AronHetLam/ATEM_tally_light_with_ESP8266 #137: The hotspot state will be unstable.

weiweike, 2024-12-25:
if  Network ssid or  password save error，WiFi is not connected,  WiFi.softAP("Tally Light setup"); The hotspot state will be unstable.Hot spots sometimes don't appear.。When hot spots can be connected，"Tally Light setup"  web page is too slow or incomplete to load.Sometimes it's normal，I used 8266-12e.

jmvb-cl, 2024-12-25:
Yeah, well, it is unstable, but with a little patience, refreshing a few times, you can go through

weiweike, 2024-12-26:
> Yeah, well, it is unstable, but with a little patience, refreshing a few times, you can go through

I modified some codes, now it is super stable.

jmvb-cl, 2024-12-26:
> > Yeah, well, it is unstable, but with a little patience, refreshing a few times, you can go through
> 
> I modified some codes, now it is super stable.

Can you share it?

weiweike, 2024-12-26:
> > > Yeah, well, it is unstable, but with a little patience, refreshing a few times, you can go through
> > 
> > 
> > I modified some codes, now it is super stable.
> 
> Can you share it?

IPAddress Ip(192, 168, 2, 1);
    IPAddress NMask(255, 255, 255, 0);
    char password[64] = "12345678";

   else if (firstRun) {
                firstRun = false;
                WiFi.mode(WIFI_AP_STA); 
↓↓↓↓↓↓↓↓↓
    WiFi.mode(WIFI_AP);
    WiFi.hostname(deviceName);
    WiFi.softAPConfig(Ip, Ip, NMask);
    WiFi.softAP(deviceName, password);

AronHetLam, 2024-12-26:
I'm aware of this, but sadly the change you made results in the ESP not being able to automatically reconnect if it loses WiFi connection.

closed 2024-12-26 by AronHetLam