AronHetLam/ATEM_tally_light_with_ESP8266 #142: ESP32 C3 Compilation error arduino IDE

regamp, 2025-04-20:
I want to load the program on an ESP32 C3 (Lolin C3 pico) but in compling it wants to load an ESP8266 library which is not needed.
Tally Version: v3.3.2
board: ESP32 C3 (Lolin C3 Pico)
I use  Fastled IO07 

The esp8266 v2.7.4 boards package is installed
ESP32 v3.2.0 

Error:

>`50 | #include <ESP8266WebServer.h>
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Compilation error: ESP8266WebServer.h: No such file or directory`

AronHetLam, 2025-04-20:
Did you select your ESP32 C3 board in the list of boards? Or did you select an esp8266?
If you select an ESP32 board the 'ESP32' variable should be defined, making it select the ESP32 libraries instead on the ESP8266 ones.
Selecting the correct board is also crucial for the ensuring the program works correctly on your specific board.

Please double check your configuration. If can't get it working, please send screenshot of your selection and IDE.

regamp, 2025-04-20:
[image]
I tried with ESP32C3 Dev module , but same error.

AronHetLam, 2025-04-20:
Odd, it looks correct to me. I'll need to look into what might have changed.

You could try to add the line
`#define ESP32=1`
At the top of the file and see if that gets you going until I can have look at it.

regamp, 2025-04-20:
[image]

AronHetLam, 2025-04-21:
I had a quick look at it this morning. Something has changed, but I haven't found a solution yet. When using an ESP32 board the 'ESP32' flag used to be set, but it isn't anymore...

Try '#define ESP32 1' instead of just '#define ESP32'.
That at least made it select the correct libraries for me.
If you run into issues with 'FastLED' try downgrading FastLED to 3.6.0

regamp, 2025-04-21:
Working with `#define ESP32 1` and FastLED 3.9.16
 I changed the `IPADDR_NONE` to `INADDR_NONE` because i got and an error "Compilation error: ambiguous overload for 'operator!=' (operand types are 'IPAddress' and 'u32_t' {aka 'long unsigned int'})"

`if (settings.staticIP && settings.tallyIP != IPADDR_NONE) {`
`if (settings.staticIP && settings.tallyIP != INADDR_NONE) {  // or IPAddress(0,0,0,0)`

I tried with FastLED 3.6.0 but didn't work for me.
Thanks for your help

closed 2025-04-21 by regamp