AronHetLam/ATEM_tally_light_with_ESP8266 #144: Not connected to the ATEM Switcher

regamp, 2025-04-26:
ESP32C3 
I changed the code to make it work: Added`#define ESP32 1` and changed `!= INADDR_NONE`
ATEM IP is correct
The program always wants to connect to 255.255.255.255 
I have tried with the testserver but the same error.
With an another program the ESP32 can ping the ATEM switcher.
[image]

AronHetLam, 2025-04-26:
Strange, I'd have to look into that.
Have you tired the web installer? The Atom matrix option should work on ESP32. Would you try that and let me know if can connect it to the ATEM?

regamp, 2025-04-26:
The web installer says that ESP32C3 is not supported. I used the arduino IDE because I had to change the LED pins too.

AronHetLam, 2025-04-26:
Ok. It's a while since I had it running last, and it looks like a few things have broken since due to new releases.

AronHetLam, 2025-04-26:
Try with the regular tally light on the web installer. I believe that supports the ESP32

regamp, 2025-04-26:
Regular version not working for me.
Serial monitor:
Invalid pins selected

AronHetLam, 2025-04-26:
I think I've found the various build issues, and have pushed the fixes.
Regarding the issue you had with the IPAddress being wrong I'm unable to reproduce it. If you still have the issue please let me know.

regamp, 2025-04-28:
I tried connecting to the test server and it won't connect.
Maybe I'm just messing something up or does not work with ESP32 C3.
[image]
[image]

AronHetLam, 2025-04-28:
It should definitely write the switcherIp in the `sending connect packet...` lines. It did in my log yesterday while testing.
Have you accidentally added additional include statements at the top of the file?

regamp, 2025-04-28:
I changed the LED1, LED2(ESP32C3 doesn't have 32,33.... pins) and FastLED Pins (to 7, onboard LED)
[ATEM_tally_light.txt](https://github.com/user-attachments/files/19946032/ATEM_tally_light.txt) my code

AronHetLam, 2025-04-28:
Don't ask why, but the bottom 4 lines in the following fixes it.

```c++
    //Read settings from EEPROM. WIFI settings are stored separately by the ESP
    EEPROM.begin(sizeof(settings)); //Needed on ESP8266 module, as EEPROM lib works a bit differently than on a regular Arduino
    EEPROM.get(0, settings);

    //Ugly fix for IPAddress not loading correctly when read from EEPROM
    settings.tallyIP = IPAddress(settings.tallyIP[0], settings.tallyIP[1], settings.tallyIP[2], settings.tallyIP[3]);
    settings.tallySubnetMask = IPAddress(settings.tallySubnetMask[0], settings.tallySubnetMask[1], settings.tallySubnetMask[2], settings.tallySubnetMask[3]);
    settings.tallyGateway = IPAddress(settings.tallyGateway[0], settings.tallyGateway[1], settings.tallyGateway[2], settings.tallyGateway[3]);
    settings.switcherIP = IPAddress(settings.switcherIP[0], settings.switcherIP[1], settings.switcherIP[2], settings.switcherIP[3]);
```

I was able to reproduce after a clean install.
For some reason read/write of IPAddress to EEPROM has changed. Later I'll see if i can find a better solution, but this at least works....

closed 2025-05-03 by AronHetLam