AronHetLam/ATEM_tally_light_with_ESP8266 #145: Not issue, turn D4, D5, D6 into ground out

jmvb-cl, 2025-05-03:
At this moment, I have my tallys turning on my camera monitors on board tally lights, adding a relay.

The monitor tally lights turn on putting to ground a couple of rj45 pins.

I saw this post, where a guy achieve turning an ESP32 GPIO pin into Ground.

I was wondering if it possible turn D4, D5, D6 into ground out?

https://www.reddit.com/r/Esphome/comments/11xnjcj/turning_a_gpio_pin_into_ground/

Best Regards

jmvb-cl, 2025-05-03:
That was fast... :(

AronHetLam, 2025-05-03:
I would start by measuring the voltage and current of those rj45 pins to ensure that you don't fry the ESP. It can only handle 3.3v and a few mA, but it's probably fine as rj45 should be within that range.
I've never used it myself, but it seems like what you're looking for is called `open drain`.
Try changing the pin modes to `OUTPUT_OPEN_DRAIN` instead of just `OUTPUT` as shown below.
```
    //Init pins for LED
    pinMode(PIN_RED1, OUTPUT_OPEN_DRAIN);
    pinMode(PIN_GREEN1, OUTPUT);
```